aikoncwd/aikoncwd-rpi-mediacenter #1: Iniciar directamente con retroPie

radikalkarrot, 2016-07-24:
Estoy buscando un sistema rapido y amigable para usar moonlight y por lo visto lo tienes muy bien montado con retroPie, por lo que he visto en tu canal de Youtube versiones anteriores no traian Kodi y se cargaba directamente retroPie, Hay alguna manera de hacer esto con la nueva version?

aikoncwd, 2016-07-25:
Sí, es muy sencillo. Arranca la imagen, cierra Kodi (opción salir) y quédate en la consola. Escribe:
    nano .bashrc

desplazate abajo del todo, verás un pequeño bloque de código, casi al final leerás el comando `kodi-standalone`. Simplemente cambia `kodi-standalone` por `emulationstation`. Guarda el fichero con Ctrl+X, Y, Enter y reinicia la raspberry con `reboot`

Si sigues los pasos correctamente tendrás Emulationstation como opción de arranque. Si te funciona avísame y cerraré la issue. Gracias!

aikoncwd, 2016-07-25:
Hola, voy a cerrar el ticket ya que esa solución que te expliqué te funcionará seguro. Si tienes problemas o no te sale, re-abre el ticket o escribe uno nuevo.

Un saludo :)

closed 2016-07-25 by aikoncwd

radikalkarrot, 2016-07-25:
Acabo de probarlo y efectivamente funciona, muchas gracias!