aikoncwd/aikoncwd-rpi-mediacenter #2: Plex y pantalla completa

nanditoelmen, 2016-07-25:
Hola en primer lugar muchas gracias por tu trabajo, tengo.una raspbrry pi 3, y de momento me han surgido 2 problemas, el primero seria que.no soy capaz de ponerlo en pantalla completa, lo que quiero decir es que tengo una franja negra en la parte de arriba de mi tv y otra el el lateral y no soy capaz de quitarlas,  lei en lo que indicaste para modificar el.overclok, tecleando /boot/config.txt pero no me sale nada al escribir eso

El segundo problema es que no consigue que me aparesca mi biblipteca de pex en Plexmbc, inicio sesion con mi cuenta o con el pin y no me aparece mi bibliotrca, porfa respondeme cuando puedas, un saludo gracias

aikoncwd, 2016-07-25:
Lo de la franja negra es sencillo de solucionar, se llama **overscan**, cierra Kodi para llegar a la consola y escribe:

    nano /boot/config.txt

Abajo del todo verás 2 variables: overscan_scale y disable_overscan. Quita el comentario (#) de delante y prueba con esos valores, si no prueba cambiando el 0 por un 1 o viceversa. Tendrás que guardar los cambios en el ficherro (Ctrl+X, Y, Intro) y reiniciar para ver los cambios.

Lo de Plex... no sabría decirte, seguramente se trate de algún fallo de configuración/puertos en tu red local de casa.

nanditoelmen, 2016-07-25:
Hola gracias por responder tan rapido, probare lo que me has indicado, y lo del plex pues me tiene rayado porque en cambiao si pruebo con rasplex funciona bien, haber si leendo.por ahi encuentro la solucion, muchas gracias , saludos

aikoncwd, 2016-07-25:
Si tengo tiempo, ésta semana me instalaré plex server en mi PC he intentaré acceder a la biblioteca desde el cliente plex-raspberry, si veo algún fallo te lo comentaré. Si te parece cerraré ésta incidencia. Si continuas con errores o con algo nuevo, puedes re-abrirla o abrir una nuevo ticket.

Un saludo! :)

closed 2016-07-25 by aikoncwd