aikoncwd/aikoncwd-rpi-mediacenter #4: Fallo autoexpandir

Kikobeats, 2016-07-25:
No me aparece 100%

```bash
root@mediacenter-v6:~ # df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G  4.4G   24G  16% /
devtmpfs        364M     0  364M   0% /dev
tmpfs           368M     0  368M   0% /dev/shm
tmpfs           368M  5.3M  363M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           368M     0  368M   0% /sys/fs/cgroup
/dev/mmcblk0p1   60M   22M   39M  36% /boot
tmpfs            74M     0   74M   0% /run/user/0
```

Kikobeats, 2016-07-25:
Supongo que esto también está relacionado:

```
root@mediacenter-v6:~ # /etc/rc.local
rmmod: ERROR: Module xpad is not currently loaded
```

Kikobeats, 2016-07-25:
Creo que lo he entendido mal, no tiene que poner 100%, sino que la partición tiene que ser el total de tu sd 😄 !

Cierro a menos que me digas lo contrario, voy probando cosas. Gracias!

closed 2016-07-25 by Kikobeats

aikoncwd, 2016-07-25:
Exacto, me expliqué mal en el manual... SI la partición no está extendida, el porcentaje de *used* marcará 100%. Si se extiende correctamente entonces marcará menos (en tu caso solo 16%) :) Quizás tenga que mejorar la explicación del readme para evitar confusiones.