aikoncwd/aikoncwd-rpi-mediacenter #5: Error En Pantalla Monitor

Crakxploit, 2016-07-31:
Error Al Iniciar En Monitor De PC A Resolución 1366x768, Se Queda Totalmente En Negro, Eh Probado Otras Versiones Tales Como Retropié y Inicia Correctamente, E' Seguido Todas Tus Indicaciones Al Pie De La Letra, Me Gustaria Saber Como Arreglo Eso, Ya Que Tu Versión De Mediacenter Es Mucho Mejor

aikoncwd, 2016-08-01:
Realiza las siguientes pruebas:
1. Vuelve a grabar la SD, es posible que se haya grabado mal
2. Prueba a conectar la RPi en otro TV o monitor, solo para comprobar si arranca correctamente

query10, 2016-08-05:
Hola, me ha pasado lo mismo con un monitor hdmi, en la tele funciona perfecto.

Muchas gracias por el aporte me has ahorrado muchas horas.

aikoncwd, 2016-08-19:
Habéis podido solucionar el problema? Os puedo ayudar en algo más?

closed 2016-09-05 by aikoncwd