aikoncwd/aikoncwd-rpi-mediacenter #10: Ayuda con pifba

elk9999, 2016-08-18:
Hola ,tengo un problema con el emulador fba, cuando juego kof para seleccionar los personajes en random,como todos saben hay q presionar el start para eso, la cosa es q apreto start por 3 segundos y entra a la configuracion de hardware del juego. Hay alguna configuracion para pifba para asignar algun hotkey o deshabilitarlo. Gracias y excelente trabajo con tu imagen, muy bien configurado todo y facil de usar.

aikoncwd, 2016-08-19:
No utilizo muy a fondo ese emulador, te dejo link oficial de la wiki son toda su información. Quizás puedas encontrar lo que buscas [aqui](https://github.com/retropie/retropie-setup/wiki/FinalBurn-Alpha).

Puedes probar a utilizar otro emulador d eneogeo en lugar de pifba, por ejemplo (GnGeo-Pi), (PiFBA), (lr-fba), (lr-fba-next). Si encuentras la solución, agradecería que me avises para ponerlo en el tutorial :)

elk9999, 2016-08-19:
Gracias apenas tenga tiempo lo reviso y comento a ver que solucion hay.Saludos desde Argentina

closed 2016-09-05 by aikoncwd