aikoncwd/aikoncwd-rpi-mediacenter #11: Problema al descomprimir Mediacenter-AikonCWD-v6.7z

Elzorro10, 2016-08-23:
Buenas...

Lo primero agradecer todo el curro.

Me he bajado el archivo torrent de la imagen y tengo problemas al descomprimirlo. 
El descompresor utilizado es el 7-Zip
Este es el mensaje de error que me sale al intentar descomprimirlo:

=============================
# Mensaje
0 F:\Mediacenter-AikonCWD-v6.7z
1 Método de compresión no válido para ´Mediacenter-AikonCWD-v6.img`
=============================

Estoy haciendo algo mal o es que el archivo le falta algo de información.

Muchas gracias!

alexdudey, 2016-08-25:
Hola
Yo tengo el mismo problema. ¿Lo has podido resolver?
¿Alguien puede resubir un torrent para poder descargar el archivo en buen estado?

Gracias!

Elzorro10, 2016-08-29:
No... no lo he podido resolver... sigo funcionando con la imagen 5 beta... a la espera de un alma caritativa....

Un saludo a todos y gracias si alguien puede subir una nueva imagen

alexdudey, 2016-08-29:
Yo lo pude resolver. El problema estaba en que lo intentaba descomprimir con winrar y es el que da error. Me descargué el programa 7zip y con él sí que lo pude descomprimir sin problema. Pruébalo y me comentas.

Elzorro10, 2016-08-29:
He actualizado a la ultima version de 7zip y ya puedo descomprimir la imagen. Perfecto.

Era una version antigua del 7zip

Un saludo

closed 2016-08-29 by Elzorro10