aikoncwd/aikoncwd-rpi-mediacenter #15: No arranca retropie despues de actualizar

Razak99, 2016-09-24:
Hola, despues de actualizar todo retropie (modulos incluidos) no me arranca. Desde kodi lo selecciono, sale pantalla negra y acaba volviendo a kodi. Es porque no se puede actualizar? O porque algo ha quedado mal? Desde la consola ejecuto emulationstation y me dice q no deberia ejecutarse como root...

Gracias por adelantado!

aikoncwd, 2016-10-11:
emulationstation tiene una restricción para evitar ser ejecutado como root. Si actualizas retropie, esa restricción se activa (por eso no te funciona). Soluciones:

- Vuelve a grabar la imagen original y no actualices
- Edita el fichero `/usr/bin/emulationstation` y elimina la segunda línea, para eliminar la restricción de root.

closed 2016-10-11 by aikoncwd

albertoner81, 2017-03-29:
Hola, lo primero, muchisimas gracias por todo el curro que te pegas.

Y lo segundo, ¿a que línea te refieres?, porque la he eliminado y no me arranca, ahora tengo retropie 4.2

Muchas gracias

alvmoran, 2017-04-03:
albertoner81 por favor, puedes indicar que archivo de emulstation y que lineas se debe eliminar.

Gracias.

albertoner81, 2017-04-03:
Edita el fichero /usr/bin/emulationstation lo pone aikoncwd unos post más arriba.

Salidos digo saludos.

alvmoran, 2017-04-03:
Gracias y perdona mi torpeza, pero en retropie 4.2 para raspberrypi 3, mediante red no veo ese fichero.

albertoner81, 2017-04-03:
Yo lo edité desde raspbian y no te preocupes yo soy más torpe que tú.

alvmoran, 2017-04-03:
Por favor, podríais idicarme las lineas a eliminar.
A ver si localizo desde conexion ssh que archivo es el que hay que editar.
Ya que no tengo usr/bin
Solo tengo parecido
Config/emulationstation y alli varios archivos pero su codigo solo hace referencia a bitones de los mandos.

albertoner81, 2017-04-03:
Pues ni idea tío, ahí me pillas. Lo siento mucho.