aikoncwd/aikoncwd-rpi-mediacenter #19: No funciona el streaming de Steam con moonlight.

lubinos, 2016-10-20:
Buenas, al intentar hacer streaming de mi biblioteca de Steam desde mi pi3, me sale en bucle el siguiente mensaje:

Waiting for IDR frame.

He buscado por Google y he visto un hilo de github que dice que se arregla en la versión 2.2.2 de moonlight-embedded.

¿Cómo puedo saber que paquete hay instalado en tu imagen?
¿Cómo puedo actualizar el paquete?

lubinos, 2016-10-20:
Se me olvido poner el enlace al hilo de github

https://github.com/irtimmer/moonlight-embedded/issues/424

lubinos, 2016-10-21:
Actualizo para comentar que solucione el problema actualizando a la versión 2.2.2, para ello desde consola ejecute los siguientes comandos:

apt-get update
apt-get install moonlight-embedded

Tras realizar la instalación solo tuve que eliminar el dispositivo desde shield en mi pc y volver a enlazarlos (pair)

aikoncwd, 2016-11-04:
@lubinos disculpa mi demora en contestar, me alegra saber que lo has solucionado. Un saludo!

closed 2016-11-04 by aikoncwd