aikoncwd/aikoncwd-rpi-mediacenter #20: Servidor VNC

moracabanas, 2016-10-29:
Me gustaría usar un servidor VNC para poder acceder por tunelacion SSH. Alguna idea para llevarlo acabo? Y por otra parte si se lleva a cabo con un daemon consumiría demasiados recursos? Gracias

moracabanas, 2016-10-29:
He probado con este tutorial pero no me funciona: 
https://www.raspberrypi.org/forums/viewtopic.php?t=123457

aikoncwd, 2016-11-04:
Ese tutorial es correcto y debería funcionar... pero para que quieres un sevidor VNC? Las aplicaciones de Kodi y RetroPie no podrán ser enviadas por VNC... si quieres controlar tu RPi de forma remota, lo mejor es que utilices el protocolo SSH (ya instalado). Si quieres acceder por VNC desde fuera de tu LAN local, recuerda abrir NAT en tu router (o usar DMZ), consulta en las instrucciones del fabricante de tu router.

closed 2016-11-04 by aikoncwd

moracabanas, 2016-12-10:
Hola gracias por la respuesta! Me interesa instalar las herramientas de seguridad desde el repositorio Kali Linux y algunas de las herramientas son muy sencillas de usar desde la interfaz gráfica. Como a veces no tengo acceso a una pantalla, es muy comodo usar VNC desde el móvil y aprovechar el potencial de raspberry pi como pen tester. El problema es que por lo visto debe haver algún bug ya que al iniciar VNCme aparece un escritorio totalmente gris y en general la conexión y el puerto son inconsistentes, y me cuesta hacerlo funcionar como un daemon para no tener que activarlo cada vez que la enciendo. Es decir actualmente hay que acceder por SSH y "encender" el acceso por VNC lo que resulta incomodo. Y por supuesto soy un negado en las configuraciones de inicio en init.d y demás

moracabanas, 2016-12-10:
Puede que sea mas sencillo prepararme un SD para ese propósito, de todas formas gracias por tu ayuda y por tu aportación, es un gran trabajo y encima en español y bien documentado. Un saludo y gracias por todo!

limonetty, 2017-01-12:
Moracabanas,
Si aun quieres corregir el problema del Vnc en gris es debido a que no detecta (carga) el Desktop para ello tienes que indicarlo en el archivo **xstartup**. normalmente en el directorio del usuario (ejemplo Centos 7,  /home/“nombre usuario”/.vnc/xstartup )
Condigo del archivo (en negrita linea a modificar)
   #!/bin/sh
   unset SESSION_MANAGER
   unset DBUS_SESSION_BUS_ADDRESS
  #exec /etc/X11/xinit/xinitrc 
  **exec /bin/sh /etc/xdg/xfce4/xinitrc** --> Modificado para cargar el Desktop Xfce

Igualmente tal como comenta Aikoncwd, no es un solución practica Vnc en el MediaCenter, ademas para el correcto funcionamiento de Kali es más que aconsejable usar un SdCar independiente.

Saludos