aikoncwd/aikoncwd-rpi-mediacenter #21: v6 Image doesn't boot

LastCryyy, 2016-11-04:
Hey, I understand spanish well but not enough to post here.

So here's my problem, i'm tried anything to get my sd card to boot with ur .img, I can't figure out how to install the mediacenter.

I formated the sd card, then I used Win32DiskImageMaker, and even Rufus in DataDump mode.

It's impossible to get my card to boot, I've tried with a Rpi3 and a Rpi2b

I never installed a Rpi system before, so I think ur file isn't involved in my issue

Thanks

aikoncwd, 2016-11-04:
Start with a full formated microSD card, ensure you have enought space to burn the img file.
Download the img file again, check the checksum to ensure you have a valid img file [here](https://github.com/aikoncwd/aikoncwd-rpi-mediacenter#hash-del-fichero-mediacenter-aikoncwd-v6img).

Follow tutorial, then boot your RPi, if you have any error, post a screenshot (or photo) of yout boot screen.

lrasmussen79, 2016-11-17:
Hi there! I have the same problem, booting from a micro SD Kingston Class 10. 
Today i try to boot from the micro SD card of my wife smartphone (shhh! don't tell her! ) and works fine! 
The only diference is that SD card is a class 4 micro SD card! 
I'm super newbie on this, but this, the SD card class 4,  work fine for me. 

Regards!

aikoncwd, 2016-11-18:
Look at [this table](elinux.org/RPi_SD_cards)

There is a list of tested SDcards for RPi, there are some brands that don't work/boot correctly :( Maybe you have one of those microSD? I recommend this microSD card [SanDisk SDSQXNE-032G-GN6AA](https://www.amazon.es/gp/product/B013CP5A0Y/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1). It's fast and it works (because I have this one)

jnca27, 2016-11-21:
Hola tengo un problema paecido, a mi si me carga la imagen del garli-dog pero se queda ahi y desúes me pone esta imagen:
tengo la raspberry pi 3
[image]

lrasmussen79, 2016-11-21:
Probaste con otra tarjeta de memoria? 
A mi me paso que con una Kingston de 16Gb Class 10 no me anduvo, pero si con una Kingston de 16Gb Class 4. 
Saludos!

jnca27, 2016-11-21:
tengo dos memorias maxwell de 8gb clase 4, creo que intentaré lo que me mencionas, ahora mismo voy a steren
gracias!!!

jnca27, 2016-11-22:
Efectivamente era la micro SD chafa que tenía jeje ya estoy disfrutando al 100% de mi consola. Una duda, para no apagar la consola en linea de comandos quisiera saber lo siguiente: cerrando kodi puedo desconectar directamente la alimentación de poder? o aún así corro el riesgo de que se dañe la micro sd?

aikoncwd, 2016-11-29:
Desde Kodi, pulsa el botón de salir y selecciona "apagar sistema". Luego ya puedes desenchufar el cable. Desde la consola simplemente escribe `poweroff`

Un saludo

closed 2016-11-29 by aikoncwd