aikoncwd/aikoncwd-rpi-mediacenter #22: Imagen aparece dañada (novato con RPi)

joakini10, 2016-11-04:
Hola, me gusta mucho tu trabajo pero no consigo instalarlo en mi raspberry pi 3. 
descargo la imagen con utorrent y me aparece una carpeta comprimida en formato .7z
Cuando digo a descomprimir me dice que el archivo está dañado. He mirado el tutorial muchas veces y no escribes nada sobre descomprimir el archivo. Es que hay que montarlo asñi como viene, en formato .7z??
Estoy muy perdido. Muchas gracias de antemano.

aikoncwd, 2016-11-07:
El formato 7z es un formato comprimido, como ZIP o RAR, para descomprimir la imagen necesitas instalar el programa [7Zip](http://www.7-zip.org/a/7z1604.exe)

Una vez descomprimas correctamente el fichero 7z obtendrás un fichero con extensión `.img`, ese fichero tendrás que grabarlo en tu microSD (sigue el tutorial). Un saludo! :)

closed 2016-11-07 by aikoncwd

joakini10, 2016-11-12:
Gracias por contestar! me habia pasado que la descarga estaba corrupta, ya lo he solucioonado. muchas gracias.

FroYsSanT, 2017-02-17:
Hola buenas!
Me pasa algo similar... me bajo la imgen, descomprimo pero a la hora de montarla con el win32diskimage se tira un monton de tiempo haciendo como que copia a la sd pero al finalizar el proceso no hace nada y si intento montar el .img me dice que el archivo esta dañado... alguna idea?

aikoncwd, 2017-02-20:
@FroYsSanT Vuelve a descargar la imagen, es posible que esté dañada. Comprueba el MD5 para asegurarte que la imagen es correcta.

FroYsSanT, 2017-02-20:
Hola @aikoncwd , ya probe a bajarme varias veces la imagen el otro dia, incluso desde ordenadores y casa diferentes, pero nada... la imagen me la da dañada.
Me la acabo de bajar de nuevo para volver a probar, me he bajado un programa para comparar el md5 como me dijiste... nada. La imagen esta dañada =(

aikoncwd, 2017-02-20:
El MD5 no te coincide? Ahí está el problema... usas 7zip para descomprimir la imagen?

FroYsSanT, 2017-02-20:
Perdona... me mal explique. Con "nada" queria decir que no hay problema jeje. Segun tu hilo en GitHub el MD5 de tu archivo es - MD5: 6AD3E6CE6CFCAB87991C8F317D9D3CF1 - y con el programa externo para windows para sacar dicha info me da este - MD5: 6ad3e6ce6cfcab87991c8f317d9d3cf1 - .Es el mismo, lo unico las letras en minuscula.
Si, uso 7zip para descomprimir. Mis pasos son...:
1º.- Me bajo el 7zip del torrent de tu hilo de GitHub.
2º.- Descomprimo con 7zip.
3.- Cargo el Mediacenter-AikonCWD-v6.img en el Win32DiskImager, selecciono d:/ (mi SD) y le doy a write.
4.- Se tira un monton de tiempo (a mi juicio algo mas de lo normal comparado con otras copias que he hecho... pero es lo de menos, que tarde lo que sea... es solo una apreciacion mia) y cuando acaba todo el proceso la SD sigue teniendo 28,9 Gigas (uso SD de 32) libres y esta completamente vacia. ¿Debo darle algun formato en especial antes de la copia? Uso el programa SDFormater para vaciar tarjetas.

Y ahi me quedo... no se me ocurre que mas hacer. Alguna sugerencia?
De nuevo, gracias de antemano por tu tiempo ;)

FroYsSanT, 2017-02-20:
@aikoncwd ! El maravilloso mundo de la informatica!!! Acabo de hacer de nuevo todo el proceso, exactamente tal cual las veces anteriores.... y ahora si ha hecho una copia correcta!!! Me han dado ganas de reventar la pi! xDDDD
Nada, muchisimas gracias por tu tiempo de nuevo, perdona las molestias y a ver esa v7 cuando cae!! Que le tengo ganas para la otra pi de mi padre! Mucho animo para seguir por el camino que llevas que vas genial !!!!!!
Un saludo!!!

FroYsSanT, 2017-02-22:
Hola  de nuevo @aikoncwd y perdona la molestia otra vez... 
Cuando tiro a emparejar con el moonlight... si solo uso el comanto moonlight pair me intenta emparejar contra una ip que no se de donde se la saca.... Esto lo he solucionado con con el comando moonlight pair 192.168.1.50 (vamos.... metiendo la ip fija de mi sobremesa) y ya meti el codigo de 4 cifras. Se supone que ya esta emparejado... pero al ejecutar steam desde retropie intenta lanzar steam desde la otra ip que no corresponde a nada de lo que yo tengo en casa... he mirado a ver si podia modificar de alguna forma la linea cuando entro a "retropie -> steam -> lanzar a 1080 60fps" para meterle manualmente la ip del sobremesa... pero nada, no he encontrado la forma.
Tengo tanto sobremesa como pi cofigurado con ip estatica. Como modifico en retropie para que linkee a la ip que yo quiera?
Gracias de nuevo de antemano!