aikoncwd/aikoncwd-rpi-mediacenter #23: Problema configurando wifi

NoSoyIgnacio, 2016-11-05:
Buenas,me acabo de iniciar tanto en raspberry como en linux y acabo de instalar esta build y estoy teniendo un problemita,cuando intento configurar el WiFi desde Kodi cambia al Desktop pero mas nada ocurre,ya modifique el dhcpcd.conf como esta en el tutorial,y ya intente modificarlo desde el icono en la barra pero al darle click derecho a lo que seria el WiFi no sale ningún menú,pero el bluetooth si funciona en ambas instancias.Estoy haciendo algo mal?

aikoncwd, 2016-11-07:
Estando en Kodi, si pulsas el icono de configurar el wifi, que te sale? Puedes adjuntar una foto para poder ver el error? Gracias

NoSoyIgnacio, 2016-11-07:
Buenas,disculpas por no actualizar el post,ayer logre accesar al wifi config,el problema era que se debe abrir el wifi config desde que Kodi bootea si se abre otra aplicacion y luego se vuelve a entrar a Kodi y se abre el wifi config lo que hara es enviarte al Desktop sin presentarte la configuracion del wifi o algun mensaje de error,o al menos es lo que sucede en mi caso.Aun no logro accesar al wifi,pero estoy seguro que es por alguna configuracion que esta mal de mi parte,estoy trabajando para resolverlo.

lrasmussen79, 2016-11-17:
Yo tengo el mismo problema. 
La pregunta es: como puedo lograr habilitar wifi, de manera que me detecte automáticamente la red, y pueda poner mi contraseña de wifi? Esto lo quiero dado que tengo pensado trasladar el equipo. 
Intente configurar editando el archivo dhcpcd.conf, editando las opciones de wlan0, pero no sucede nada... 
Luego de editar, hay que reiniciar? (esto no lo hice).
Por ultimo, hay manera de activar DHCP? Habia encontrado algo, pero quedo en mi laptop... Ahora estoy en el trabajo. 
Saludos y gracias!

aikoncwd, 2016-11-18:
@lrasmussen79 Simplemente instala la RPi en la nueva ubicación y arranca el sistema. Estarás en Kodi, luego utiliza los accesos directos del menú **programas** para acceder al **wifi config**, desde ahí podrás escanear las redes visibles y conectarte sin problemas.

lrasmussen79, 2016-11-18:
Gracias! El problema es que ni en mi casa me funciona con Wifi. 
Edite dhcpcd.conf, poniendo una ip fija, y en ningun momento me da la opcion de conectarme a mi wifi. 
Por otro lado, realice esto que mencionan en el post #817: 
http://www.mediavida.com/foro/hard-soft/raspberrypi-v2-montando-mejor-mediacenter-546612/28

Pero no me ha funcionado, esta noche con tiempo seguire intentando. 

Cualquier sugerencia es bienvenida. 
Saludos!

aikoncwd, 2016-11-29:
Sigue el tutorial del [readme oficial ](https://github.com/aikoncwd/aikoncwd-rpi-mediacenter/blob/master/README.md), ahí se explica paso por paso como configurar una IP estática.

closed 2016-11-29 by aikoncwd