aikoncwd/aikoncwd-rpi-mediacenter #32: [Sugerencia] Añadir link en Mega

ismael-dm, 2017-01-06:
Al intentar descargar la imagen, la velocidad de descarga es extremadamente lenta debido a la reducida cantidad de semillas ("compis"), lo que alarga la descarga hasta un tiempo de incluso 5 días.
Subiendo el archivo a Mega (o similares) el tiempo se podría llegar a reducir a minutos.

Dejhavi, 2017-01-06:
Yo he descargado a 1Mb/s con 59 semillas y 21 pares. Solo ha tardado 20 minutos

aikoncwd, 2017-01-09:
He comprobado el torrent desde otro equipo, no he tenido ningún problema en descargarlo a una velocidad aceptable (aprox 15 minutos). Revisa la configuración de tu cliente torrent. Si algún usuario quiere subir link a Mega, que lo haga.

closed 2017-01-09 by aikoncwd