aikoncwd/aikoncwd-rpi-mediacenter #34: Update Retropie

limonetty, 2017-01-11:
Buenas noches,

Muy buen trabajo con la imagen. Gracias!!!
He montando la imagen AikonCWD-v6 y he provado de actualizar el S.O y el 
Retropie.
El S.O lo he actualizado de la siguiente manera
apt-get update
apt-get upgrade
Reboot 
     todo funciona ok
apt-get dist -upgrade
reboot
      Todo funciona ok
Actualizo Retropi mediante el entorno gráfico de emulestation menu (ejecuta el script)
reboot
Retropie no inicia ni desde el icono del kodi ni desde terminal muestra el siguiente error :

emulationstation should not be run as root. If you used 'sudo emulationstation' please run without sudo.

Imagino que es un problema del user? alguna pista? Es un problema conocido?
Version kodi: 16.1
Last Version Retropi: 4.0 
Current Version Retropie: 4.1.9 
Gracias por la ayuda

aikoncwd, 2017-01-12:
Hola @limonetty 

Exacto, es un problema conocido. La instalación original de retropie/emulationstation está modificada para que pueda ser ejecutada como root (el usuario default de ésta imagen). Si actualizas el sistema se machacarán esos cambios y retropie no se podrá iniciar:

> emulationstation should not be run as root. If you used 'sudo emulationstation' please run without sudo.

Solución? Lo más rápido es que vuelvas a grabar la imagen original y **no** actualices. Otra opción sería explicarte las modificaciones que he realizado en emulationststaion y tú mismo los apliques a tu versión actualizada. Lo que pasa es que dichos cambios no me los se de memoria y tendrías que esperar unos días hasta que pueda responderte con los pasos.

Que prefieres hacer?

sbarri, 2017-01-12:
Buenas, 

He tenido el mismo problema que @limonetty. Yo te agradecería si pudieses comentar la configuración y los cambios a realizar sobre la imagen, no por este hecho puntual, sino por no molestar en las distintas actualizaciones que se pueden realizar del sistema en el futuro.

Pido perdón si molesta mi opinión o ensucio la duda.

Muchas gracias por el curro @aikoncwd 

Un saludo

limonetty, 2017-01-12:
Buenos días Aikoncwd

Gracias por la rápida respuesta.
Tal como comenta Sbarri, seria la mejor opción facilitar la configuración para poder ir actualizando el Soft. de manera independiente.
No me corre prisa, lo cierto que es para el MediaCenter del hogar puedo estar días o semanas sin el Retropie,

Muchas gracias

aikoncwd, 2017-01-12:
@sbarri @limonetty Sin problemas, hoy intento buscar un hueco en mi agenda y miro de escribir un tutorial. Las configuraciones son muy sencillas, se hacen desde la propia raspberry usando el comando/programa `nano`. A la tarde/noche actualizo con más info

sbarri, 2017-01-12:
Genial @aikoncwd muchas gracias!!

Ntham, 2017-01-14:
Otro aquí a la espera del tutorial :)
Gracias.

zemene1, 2017-01-22:
¿Se podría actualizar sólo el navegador FireFox sin que afecte a nada?. ¿Cómo se haría?. Gracias de antemano.

Grondoval, 2017-01-25:
Yo también actualicé RetroPie/EmulationStation y no me arrancaba, el autor del sistema puso una solución al problema de que no arrancaba

Al primer intento desistí de intentarlo y formatee la sd y poniendo de nuevo la imagen volviendo a tener que configurar todo de nuevo.

Pero luego probé la otra solución que aportaba que es:

Desde la Terminal que tiene Desktop: sudo nano usr/bin/emulationstation y borras la segunda linea que indica algo de sudo o permisos de administrador.
desde el if hasta el fi. Es decir

**if
**blabla emulationstation should not be run as root. If you used 'sudo emulationstation' please run without sudo. blablabla**
fi**

Ese bloque de la segunda línea se borra, pulsas Ctrl+O para guardar los cambios (si quieres haz una copia antes por si te equivocas) y luego Ctrl+X para salir del documento, se hace un sudo reboot desde la terminal o se apaga el sistema y se inicia de nuevo y ya debería de arrancar el emulationstation.

aikoncwd, 2017-01-26:
Correcto @Grondoval, esa es la solución:

1. Editar el fichero de emulationstation: `nano /usr/bin/emulationstation`
2. Localizar el bloque *if ... fi* que habla sobre sudo/root
3. Eliminar todo ese bloque pulsando Ctrl+K, nos aseguramos que quitamos el `exit 0`
4. Guardamos los cambios con Ctrl+x, Enter, Y

Cierro la incidencia, gracias por el aporte.

closed 2017-01-26 by aikoncwd

limonetty, 2017-01-26:
Gracias! funciona perfectamente.
Un saludo