aikoncwd/aikoncwd-rpi-mediacenter #35: Imposible acceder por Samba

phyron, 2017-01-12:
Buenas!

Primero darte las gracias por este pedazo de paquete! ;)

Intento conectar por samba, pero no funciona. Pongo la direccion IP y me carga file://192.168.1.123/. Dandome el error de archivo no encontrado. Lo mismo me pasa con //mediacenter-v6/.

Tambien he probado de entrar por Putty, pongo la IP y se abre la consola pidiendome login y pass, una vez lo pongo, no hace nada mas. se queda en la consola quieto. He encontrado el error en un foro, pero no le dan solución:
[https://www.raspberrypi.org/forums/view ... 8&t=138631](https://www.raspberrypi.org/forums/view ... 8&t=138631)
Si pongo la Ip en el navegador, se me habre una pagina de codi con el menu de profile, remote....

Uso la raspberry 3 B, la tengo conectada por Wifi.

Gracias de antemano! ;)

aikoncwd, 2017-01-12:
Utiliza el comando `ifconfig` o `iwconfig` para cerciorarte 100% cual es la IP de tu raspberry. Luego desde tu PC deberías poder acceder a la ruta `\\192.168.1.123` (o la IP que hayas obtenido con los comando anteriores).  

Comentas también que no puedes acceder por SSH? Eso es muy raro, quizás la imagen no se ha grabado bien en tu SD. Formatea, baja la imagen de nuevo (comprueba el MD5) y vuelve a empezar. el SSH debería funcionar sí o sí

aikoncwd, 2017-01-24:
Cierro la issue ya, no hay más solución que la que se ha comentado, formateando la SD debería funcionar a muy malas...

closed 2017-01-24 by aikoncwd