aikoncwd/aikoncwd-rpi-mediacenter #39: Partición storage en usb

ronezgz, 2017-01-19:
Buenas tardes,

lo primero darte las gracias por esta pasada de imagen que has hecho, que entiendo lleva curro hasta decir basta. Y ahora mi duda, me gustaría que en mi raspberry pi 3 arrancara desde la micro SD, pero luego toda la instalación estuviera en un usb, antes en la raspberry pi 1 tenía openelec que al ser mas sencillo, solo había que modificar el cmdline.txt, pero ahora no tengo ni idea de como hacerlo, si me pudieras echar una mano te lo agradecería.

Muchas gracias y un abrazote

aikoncwd, 2017-01-20:
No lo he probado, pero creo que es el mismo procedimiento. Copiar el contenido de la SD al USB y luego editar el cmdline.txt, voy a buscar info por internet, pero ésto es algo que yo jamás utilizo...

ronezgz, 2017-01-20:
En openelec había que sustituir “disk=/dev/mmcblk0p2” por "disk=LABEL=STORAGE" y listo por si te sirve de ayuda, y como es obvio copiar la partición ext4 de la sd al pen

aikoncwd, 2017-01-24:
No puedo ayudarte, desconozco el procedimiento para hacer eso. Tampoco conozco las ventajas de utilizar el USB en lugar de la SDcard. He visto que con el nuevo firmware de RPi3 han habilitado el boot por USB, te dejo el tutorial completo por si quieres probarlo...  

https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md

En [éste hilo](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=155167) discuten sobre la nueva opción, quizás puedas resolver dudas sobre ese proceso. Un saludo :)

closed 2017-01-24 by aikoncwd