aikoncwd/aikoncwd-rpi-mediacenter #40: Google Chrome y Mando compatible inalámbrico xbox360

zemene1, 2017-01-21:
Hola. He descargado tu imagen y es fabulosa: Es todo lo que necesito. Enhorabuena por tu trabajo.
Mi consulta es triple:

1º - Necesito descargar e instalar Google Chrome como navegador web para poder ver Amazon Video. ¿Cómo lo hago?. Antes he intentado hacerlo desde Kodi pero nada de nada. Espero tu ayuda.

2º - Tengo un mando inalámbrico compatible con xbox360. No hay narices que MAME en retropie lo reconozca siquiera. ¿Que debo hacer?

3º - Tampoco tengo narices que Retropie me reconozca las roms de Psx. ¿Que debo hacer?.

Gracias por todo de antemano

aikoncwd, 2017-01-24:
1- La Raspberry es una solución bastante mediocre para la exploración web. Navegadores tipo Chrome o Firefox funcionan muy mal, la mayoría de contenido Java/Flash/HTML5 falla, funciona lento o directamente no funciona. Dudo mucho que puedas ver con fluidez el contenido de ese Amazon Video... de todas formas si quieres probarlo, instala chrome usando el siguiente comando en la consola: `sudo apt-get install chromium-browser`. De todas formas, si deseas visualizar Amazon Video podrías dedicar unos minutos en buscar algún addon para Kodi, yo he encontrado ésta información en google en menos de 2 minutos:  

https://www.tvaddons.ag/kodi-addons/show/plugin.video.prime_instant/
https://www.kodimania.com/viewtopic.php?t=8567

2- El mando de xbox360 te funciona en los otros emuladores? Entonces solo te falta configurarlo correctamente en MAME. MAME es un emulador especial, para configurar el joystick necesitas entrar en el menú pulsando la tecla TAB (necesitarás un teclado), te dejo [un videotutorial de youtube](https://www.youtube.com/watch?v=dUmFD92y4gU)

3- Podrías empezar por leer un poco [la wiki](https://github.com/RetroPie/RetroPie-Setup/wiki/Playstation-1), verás que para ejecutar roms de PSX necesitas la BIOS correcta. En ese link que te he puesto tienes toda la info.

Un saludo

closed 2017-01-24 by aikoncwd

zemene1, 2017-01-24:
Hola. Gracias por tu respuesta. Te puedo garantizar que antes de preguntar,
he mirado y remirado en Google.

Con respecto a la instalacion de Chrome, simplemente no funciona. Con
respecto a Kodi, ya tenía instalado ese repositorio que me aconsejas y en
el link de Kodimanía, si lo sigues, verás que a todo el mundo le pasa lo
mismo. Te logueas y nada de nada. Lógicamente, no tienes la culpa que
Amazon Prime Video no se vea en Kodi (faltaría más). Solamente buscaba una
solución alternativa que me pudieras aconsejar. Es por eso que pregunto:
¿se podría actualizar la versión de FireFox que viene en la imagen sin que
ésta se vea afectada?

Con respecto al mando compatible, había mirado un tuto erróneo. El que me
has adjuntado funciona a las mil maravillas. Gracias.

Con respecto a la compatibilidad de juegos Psx, tambien solucionado.
Gracias de nuevo

aikoncwd, 2017-01-24:
@zemene1 Puedes probar de actualizar firefox usando el siguiente comando: `sudo apt-get install firefox-esr-l10n-es-es` o quizás `sudo apt-get upgrade firefox-esr-l10n-es-es`. Pero ya te adelanto que la visualización de Youtube es pésima... no me quiero ni imaginar como funcionará Amazon Video ahí xD

zemene1, 2017-01-24:
Probaré esta tarde y ya te contaré. Gracias

zemene1, 2017-01-24:
Hola. Lo he probado y nada de nada: se queda en la versión 45 de FireFox.