aikoncwd/aikoncwd-rpi-mediacenter #42: ¿Navegador de Desktop no funciona?

Grondoval, 2017-01-25:
Inicio la Raspberry, arranca Kodi, pincho en Programas, selecciono Desktop y ejecuto el Navegador

Al pincharle no reacciona o no se abre. Tengo la ip correctamente asignada ya que puedo descargar paquetes al hacer sudo aptitude update o sudo apt-get upgrade/update.

¿Que le puede ocurrir al navegador de internet?

Grondoval, 2017-01-25:
He ejecutado firefox desde terminal con exito. poniendo firefox y se abre correctamente. Solucionado. He creado un launcher en mi escritorio hacia firefox y también funciona.

closed 2017-01-25 by Grondoval

aikoncwd, 2017-01-26:
Esa opción funciona out-of-the-box, es posible que hayas actualizado el sistema antes de probar eso? De todas formas me alegra saber que lo has slucionado