aikoncwd/aikoncwd-rpi-mediacenter #45: Al intentar configurar la Wifi desde Kodi me aparece lo siguiente?

Grondoval, 2017-01-25:
Al intentar configurar la Wifi desde kodi: Kodi -> Programas -> Wifi Config y pinchar sobre ello.

Me lleva a una pantalla de RetroPie en la que me dice como deseo configurar el steamcontroller.

¿Como puedo configurar la Wifi desde Kodi, hay alguna otra forma de configurar la Wifi?

Grondoval, 2017-01-25:
He conseguido configurar la wifi de forma manual desde la terminal usando el comando sudo nano /etc/network/interfaces

y añadiendo de forma manual mi ssid y contraseña siguiendo este tutorial:

https://geekytheory.com/tutorial-raspberry-pi-configurar-wif

Aún así creo que es un problema que no se pueda hacer un poco mas automático desde kodi al ocurrirme aún el problema del primer mensaje.

aikoncwd, 2017-01-26:
Lo que te ocurre es lo siguiente... Has actualizado el sistema con un `apt-get update, upgrade, etc...` o has actualizado el paquete RetroPie. Al hacerlo, algunos menus cambian de orden, por eso el menu de configurar wifi se mueve de lugar y terminas entrando en otra opción (por ejemplo en configurar el steamcontroller).

Desde el instalador de retropie: `./RetroPie-Setup/retropie_setup.sh` podrás llegar al menu de configurar el wifi. Intentaré añadir un aviso en la proxima versión para que no caigáis en el error de actualizar el sistema y perder esas funcionalidades

closed 2017-01-26 by aikoncwd