aikoncwd/aikoncwd-rpi-mediacenter #49: Cambios para V7

aikoncwd, 2017-02-07:
- [ ] Kodi Krypton v17
- [ ] RetroPie v 4.1
- [ ] Raspbian with Pixel Desktop
- [ ] Moonlight 2.2.3
- [ ] Transmission torrent
- [ ] PVR Simple Client IPTV
- [ ] [Ambilight/Hyperion](https://hyperion-project.org/wiki/Main)
- [ ] Spotify
- [ ] [Plex client oficial 0.0.97](https://www.kodimania.com/viewtopic.php?p=73330#p73330)
- [ ] Decine
- [ ] Exodus
- [ ] Quasar + Burst
- [ ] [Youtube addon](https://github.com/jdf76/plugin.video.youtube)
- [ ] [Movistar TV](http://www.mediavida.com/foro/hard-soft/raspberrypi-mediacenter-juegos-torrents-564777/61#1806) ?
- [ ] Twitch TV
- [ ] Popcorn Time

erpepeillo, 2017-02-07:
Yo voto por Spotify + Youtube.

Y gracias por el trabajo realizado.

JoShiaN82, 2017-02-07:
A mi me gustaria tener el Spotify  y ambiligth/hyperion 

Gracias por el tiempo que le dedicas, que a la vista está, los frutos de todo el curro.

tachademetal, 2017-02-07:
Yo voto por ambiligth/hyperion, muchas gracias por tu trabajo

marianocortijo, 2017-02-07:
Spotify, Movistar TV.
Ademas si pudieras poner Yomvi estaría genial.
Un Saludo.

josulf, 2017-02-07:
De los que hay en la v6, echo en falta el plexus streams

ericcorporation, 2017-02-07:
Spotify, Netflix y Amazon Prime Video a poder ser.
Increible Aikoncwd, excelente trabajo. Muchas gracias.

sergytheonly, 2017-02-07:
**Youtube**
**Twitch TV**
**Plex**
**Exodus**
**pelisalacarta**
**Quasar** o el que toque
**adryan**
**Catoal**
**Retropie y moonling**
Ambilight/Hyperion creo que es algo complejo, así que puede que venga bien que lo tengas preconfigurado para el resto.

El resto de addons, si son de fácil instalación e incluso alguno de los que he puesto, pueden ir fuera en otro paquete que puedas llegar a hacer en plan, addons opcional o algo así. para no sobre cargar la imagen y más si son addons de servicios de pago/premium del cual no todos disponemos.

Y luego ya te comenté que todos los providers, o funciones que no valgan para nada deberían ser purgadas, para facilitar el uso al usuario y evitar posibles malos funcionamientos. 

Addons de subtitulos estaría muy guay.

Drivers para los mandos de ps3, ps4, xbox etc, sobre todo si van por bluetooth.

El acceso directo desde Kodi a Retropie, al igual moonlight desde retropie que ya lo hiciste.

Y luego ya el resto que exprima como quiera

aikoncwd, 2017-02-07:
Gracias a todos por los comentarios, no es necesario votar los addons. Ese listado es lo que quiero añadir en la próxima V7. Si está marcado el *check* significa que ya lo he añadido. Si el addon está en la lista pero no está marcado significa que lo quiero añadir pero aún no lo he hecho o me falta configurarlo para que no de errores.

erpepeillo, 2017-02-07:
Ahhhhh.... Vale.
Bueno, lo que tu hagas sera y estara perfecto... Muchas Gracias de nuevo.
Ademas, lo que estoy aprendiendo de comandos LINUX, no est apagado..
Gracias.

dURDENN, 2017-02-07:
**Youtube** metele como el anterior lo de la API venia una por defecto metida aunque deshabilitada, asi si lo activabas luego no daba fallo a la larga que no dejaba ver videos o daba algun error. 

TwitchTV si puedes tambien incluyelo que el addon va de lujo, si no en el repositorio de kodi esta para instalarlo cada uno por su cuenta.

Y movistarTV yo le meteria que le he echado un ojo y tengo curiosidad por saber que tal va jejeje.

aikoncwd, 2017-02-07:
MovistarTV seguramente no lo dejaré instalado, pero añadiré en el manual los pasos sencillos para quien quiera usarlo. No todo el mundo tiene contratado Movistar en su casa, por eso en el listado le he puesto un **?** al lado.

ismael-dm, 2017-02-07:
Apoyo la opinión de sergytheonly, de quitar providers que no valen la pena tenerlos para facilitar el uso y evitar problemas.
También añadir drivers para mandos de ps3,ps4...

docout, 2017-02-07:
Lo primero, **Gracias por tu trabajo** aikoncwd!!
Espero con ansia esta nueva versión!!

Por último una sugerencia, si pudieras poner en Kodi el tema [Eminence 2 MOD](http://forum.kodi.tv/showthread.php?tid=246430) sería maravilloso.

josulf, 2017-02-07:
Por cierto @aikoncwd en las versiones anteriores que he probado (v5 y v6), se menciona que hay un proveedor de subtitulos para tusubtitulo.com, pero nunca lo he logrado encontrar o usar. Podrías revisar esto o decir si funcionaba en la v5/v6?

aikoncwd, 2017-02-07:
@docout Leo que esa skin esta descontinuada... la probaré a ver que tal va, pero normalmente lo más fluido suele ser la default. Cabe destacar que la default es muy versatil y permite personalizar el color de Kodi, botones y tal. Ya veré que hago :)

@josulf Lo añado por si las moscas.

jmac1311, 2017-02-07:
Ya se que prefieres no incluir en la imagen Netflix y Amazon Prime Video, pero ¿podrías indicar en el manual los pasos para poder añadirlos?

Muchas gracias por tu trabajo

flores9, 2017-02-07:
Hola @aikoncwd me gustaría a ser posible que incluyera el de Google Play Music

Chev1, 2017-02-07:
Considero que, Netflix, Amazon, Movistar TV cosas asi más particulares, no las incluiría en la de serie si quieres hacer un manual para quien los use o tenga cuenta los incorpore perfecto.

aikoncwd, 2017-02-07:
Lo dle manual lo tenía pensado @flores9 👍

jorgeleon79, 2017-02-07:
Buenos días Aikoncwd
Gracias por escuchar nuestras sugerencias.
Te hago dos por si las quieres tener en cuenta:

- Realizar dos imágenes distintas: una con retropie y otra sin retropie
- Dejar kodi los más limpio posible: como bien dices, cada uno tenemos nuestras prioridades y gustos y yo dejaría lo esencial para que la imagen este limpia y vaya fluida

Y una consulta, he leído aquí y en otro foro en el que estás que existian addons para netflix, hbo, amazon video...tenía entendido que no era posible¿? Lo digo porque tengo cuentas en las tres plataformas y me interesaría saberlo (lo montaría yo por mi cuenta)

Gracias de nuevo.
Un saludo

dURDENN, 2017-02-07:
@jorgeleon79 para eso no seria mas facil bajarte por tu cuenta la imagen oficial de Kodi desde su pagina web? Si lo quieres tal cual limpio es la mejor opcion

jorgeleon79, 2017-02-07:
@dURDENN solo es una sugerencia, creo que es lo que esta pidiendo @aikoncwd .
Agradezco tu sugerencia ,cuando tenga tiempo y mas conocimientos lo haré.

joanpi60, 2017-02-07:
Mis opciones son incluir Plex addon, limpiar providers..y no meter por defecto aplicaciones como netflix, movistar....pero sí que estén ahí para poder ser instaladas de forma fácil.

Como todo esto, ya es lo que tienes en mente...pues solo agradecerte el trabajo de ir confeccionando estas imágenes .

saludos

ivcausan, 2017-02-07:
Plex

aikoncwd, 2017-02-08:
Actualizada la lista principal, por secciones. Voy avanzando poquito a poquito. Aún estamos a tiempo de seguir agregando/quitando cosillas. Seguid participando! 😉

sergytheonly, 2017-02-08:
Plex!

dacmail, 2017-02-08:
Primero de todo muchas gracias @aikoncwd por el currazo y tiempo que nos ahorras!

Perdona mi ignorancia pero pelis/tvalacarta, popcorntime, quasar, decine, exudus, etc. no hacen prácticamente lo mismo? Hay algún motivo por el que incluir todos por defecto?

En la v6 yo al menos echaba en falta una configuración básica para descargar subtítulos, es fácil configurar pero no estaría de más que viniese incluida.

Por último, no se si es posible hacer funcionar los mandos de Xbox One (bt o cable), yo no lo he conseguido con la v6. :)

Gracias de nuevo!

P.D. Me parece perfecto dejar "fuera" a los servicios premium, aunque si supiéramos cómo hacer funcionar Netflix sería la caña!

jneivil, 2017-02-08:
Cuanto te lleva este proceso hasta finalizar la v7 ?

aikoncwd, 2017-02-08:
@dacmail Sí, más o menos esos addons hacen lo mismo, pero de diferente manera. Exodus por ejemplo mostrará contenido en VO (con posibilidad de subtítulos) mientras que el resto mostrarán contenido traducido al castellano. Pelisalacarta tiene más enlaces para torrent, etc... A más variedad de tools mejor, cada usuario tiene sus preferencias.

Lo de los subtítulos lo mejoraré en la V7 y ampliaré el tutorial explicando (con fotos) como activar los subtítulos.

En la nueva V7 voy a eliminar el driver **xboxdrv** y usaré el *actualizado* xpad-driver. Al parecer es más compatible incluso con XboxOne.

Sobre netflix... lo veo complicadísimo. No hay addon oficial por culpa del DRM y tienes que pasar por chrome+flash. Esa combinación es mortal para la RPi ya que toda la decodificación ha de pasar por CPU, provocando que vaya lento y mal. Dejo el [tutorial de netflix para RPi](https://gist.github.com/jesstelford/eded5fe8c2611f332648), verás que es bastante complicado y hay que instalar un montón de dependencias y versiones parcheadas para hacerlo funcionar *lento y mal*. Definitivamente no lo voy a incluir en la V7

@jneivil No sabría decirte, todo depende de los errores o problemas que me vayan saliendo. Un fin de semana completo no me lo quita nadie.

josulf, 2017-02-08:
Lo de los subtitulos perfecto, yo acabé instalando provider de subdivx.com porque no veía como activar los que se supone que estaban instalados.

Respecto al nuevo xpad-driver, funcionan los mandos de Xbox 360 inalámbricos?

Otra cuestión que no sé si para esta versión te lo planteas, pero estaría bien tenerlo en cuenta: no usar el usuario root para todo.

aikoncwd, 2017-02-08:
@josulf Sí que funcionan (al menos eso me consta), mis mandos son por USB y no podré probarlo. Usar una cuenta que no sea root me complicaría muchísimo el montaje de la imagen. Lo intenté allá en la V5 y muchos módulos/programas no conseguía hacerlo funcionar, problemas de permisos al grabar los torrents, errores.. un caos! Seguiré usando root como mainuser, si alguien con más conocimientos de Linux se anima, puede crear un usuario local y pelearse con los permisos y grupos de Linux, yo por el momento no es algo que me apetezca hacer 😈

ftorradoc01, 2017-02-08:
Buenas, yo sólo comentaros que usando tvalacarta, en la versión v6 los contenidos de mitele no son accesibles. Tengo un fichero que bajé que lo endiñas dentro de la carpeta del add-on y lo arreglas. No sé si en la nueva versión estará arreglado, pero en Jarvis de pc también hay que arreglarlo.

Si necesitáis el archivo lo comentáis y lo subo aquí.

Salu2

chapete86, 2017-02-08:
Grandisimo trabajo aikon. Yo añadiría mas proovedores para los subtitulos. Entiendo que el addon IPTV Simple Client, se pueden ejecutar listas locales m3u? Yo actualmente uso el playlistloader. ¿Sabes si en la nueva version de kodi estara incluido el addon de vu+?

aikoncwd, 2017-02-08:
@ftorradoc01 subelo aquí, puedes adjuntar ficheros con solo arrastrarlo a la conversación. Muchas gracias! :)
@chapete86 Sí, puedes ejecutar listas m3u locales y remotas. cual es el addon de vu+? No lo conozco

chapete86, 2017-02-08:
Es un addon PVR para decodificadores Vu+

ftorradoc01, 2017-02-08:
En este enlace está el link y las instrucciones de como añadirlo, 
http://www.mimediacenter.info/foro/viewtopic.php?f=25&t=7943
de donde lo he sacado

Este es el enlace original que hay en dicha página
http://www.mimediacenter.info/foro/download/file.php?id=1344&sid=1393b654ae16e1d56cbfb255870d50da

Lo de dentro del zip lo metes en  las carpetas kODI/ADDONS/PLUGIN.VIDEO.TVALACARTA y tambien en KODI/USERDATA/ADDON_DATA/PLUGIN.VIDEO.TVALACARTA
(creo que con lo segundo vale pero yo lo suelo pegar en los sitios)

y voilá ya fulula mitele

Muchas gracias por tu trabajo @aikoncwd 

salu2

dan83sama, 2017-02-08:
Yo añadiria Vnc server

ArdiRoberts, 2017-02-08:
yo soy un clasico y uso muchisimo plex ademas que la forma de ordenacion es cojonuda :)

Dustman71, 2017-02-08:
yo añadiría alguno de los juegos nativos de raspberry pi:

http://malagaoriginal.blogspot.com.es/2016/06/los-mejores-juegos-nativos-para.html

rbf0069, 2017-02-09:
Hola!

Hay alguna opción de que metas la configuración para el Gamepad de Xiaomi ?? Es que por más que quise en la anterior vez no había forma humana de que se configurase, solo aceptaba los cursores y con suerte el boton A, el resto nada.

Gracias!

aikoncwd, 2017-02-09:
@rbf0069 Podré poner el driver y la configuración por defecto, pero al no disponer de un mando Xiaomi no lo podré comprobar... En mi casa solo dispongo de mando SNESusb y xbox360usb. El resto de gamepads como PS3/4 o Xiaomi no tengo modo alguno de comprobarlos 😢

FlixHz, 2017-02-09:
Buenas,

Lo primero agradecerte el trabajo que haces, es estupendo. Lo segundo, si se me permite, sería sugerir incorporar unos addons, aunque si no se puede no pasa nada, pero me parecieron interesantes. 
Cine: https://www.kodimania.com/viewtopic.php?t=8463
La Liga: https://www.kodimania.com/viewtopic.php?t=7960
Y el de Arenavision EZY. Aunque no se si estos dos últimos vienen incorporados en el de Catoal o Adry, no estoy seguro, normalmente no me meto en esos dos y los addons que digo los tengo en el menú normal de Kodi.

Saludos

JuanMiguelML, 2017-02-09:
Hola @aikoncwd . 

Lo primero quería agradecerte el curro que te pegas para que nosotros podamos exprimir más aún la Raspberry Pi. 

Lo segundo, quería preguntarte, ¿esta nueva V7 servirá igualmente para la Pi 2B?

Y aprovechando que el Pisuerga pasa por Valladolid, estaría genial, como dice @rbf0069 , que pudieras añadir el driver del Xiaomi Gamepad, aunque no puedas testearlo. De eso nos encargaremos nosotros.

¡Gracias mil nuevamente!

aikoncwd, 2017-02-09:
@FlixHz Añadidos los 2 addons al listado, si me pasas el link para Arenavision EZY lo meto también.
@JuanMiguelML Sí, compatible con Pi2 y Pi3. El driver lo añadiré, junto a su tutorial para configurarlo.

thorhec, 2017-02-09:
Que ganas de que salga!
La V6 iba como un tiro!!! Seguro que está también!

En retropie vas a meter las BIOS?

¿para cuando crees que la tendrás finalizada? 

Si necesitas probadores! yo estaría dispuesto!

aikoncwd, 2017-02-09:
@thorhec Por temas legales, retropie no vendrá ni con roms ni con bios pre-instaladas. Éste fin de semana empezaré a empaquetar todo, aún me falta probar algunos addons. Supongo que en 10 días la tendré.

FlixHz, 2017-02-09:
@aikoncwd Muchas gracias aikon, te dejo el addon de Arenavision EZY www.mediafire.com/file/s4l8i7od7xi7iwf/plugin.video.arenavisionezy-1.1.4.zip

Acogros, 2017-02-09:
@aikoncwd muchas gracias por el curro q te estas pegando!
Para la nueva version podrias meter el addon de Tvheadend? Otra pregunta, no se si a al gente le pasa, pero en retropie no he conseguido cargar ninguna rom de gba, se pone la pantalla en negro como si se fuese a iniciar y se sale al menu de roms. Las roms funcionan porque en windows las he probado y funcionan, no se si sera problema del emulador.

Saludos y gracias de nuevo!

aikoncwd, 2017-02-10:
Hola @Acogros (me encanta tu avatar 😸). Miraré lo de TVheadend, si veo que no es situacional lo agregaré al listado de addons. Sobre los juegos GBA... recuerda que ese emulador requiere una BIOS instalada para poder emular esas roms. La BIOS es `gba_bios.bin`, no creo que te cueste mucho encontrarla por internet. La BIOS correcta de GBA debería tener el siguiente MD5: `a860e8c0b6d573d191e4ec7db1b1e4f6`

Acogros, 2017-02-10:
Gracias @aikoncwd (el tuyo tambien esta genial :smile_cat: ) No habia caido en lo de la BIOS. Pensaba que para los de gba no hacia falta. es facil añadirla? lo miro por ahi. Sobre el addon si puedes añadirlo estupendo y sino ya nos diras como ponerlo, que he intentado ponerlo en la V6 y no lo he conseguido....

Si necesitas a gente pa probar la nueva version, cuenta conmigo :-)

aikoncwd, 2017-02-10:
Simplemente cierra retropie/emulationstation y copia la bios en la carpeta `/root/RetroPie/BIOS`, tienes toda la info [aquí](https://github.com/retropie/retropie-setup/wiki/Game-Boy-Advance)

chapete86, 2017-02-10:
Alguien a probado a cargar las rooms desde una unidad de red compartida nfs o samba?

ivcausan, 2017-02-10:
yo las subo con filezilla. con samba también lo probé y sin problemas

ftorradoc01, 2017-02-10:
Yo he probado nfs desde un disco symbology y sin problemas.

chapete86, 2017-02-10:
me interesa por nfs en el synologic, tienes a mano el tutorial. A lo mejor es interesante incluirlo en la v7 esa parte del tutorial.

ftorradoc01, 2017-02-10:
tengo un synology 215j (de lo más baratito que hay con dos discos y va genial) lo enchufo con cable al router, asi como la rasp y lo reconoce al pelo. De scratcher le pongo el filmaffinity y así me saca las pelis con sus puntuaciones y todo ok.
El so de synology es simple pero estable y muy facil de manejar. Sólo tienes que compartir unidades por nfs y luego las encuentras con la rasp enseguida (bueno, dependiendo del router, puede tardar más).
Las roms y archivos las cargo directamente de las unidades compartidas del synology usando nfs.
Tuve que fijar las ip del synology, eso si, porque en algún reinicio de router y tal me asigno otra ip y tenía que reconfigurar en la rasp la dirección y luego me salían las pelis dobles (por no borrar los datos)

no soy un experto en synology pero llevo un par de años y he bicheao un montón. Así que si puedo ayudar en algo contad con mi hacha... :p

KurroCantos, 2017-02-11:
Hola! Sobre AirPlay o sobre la conexión bluetooth para reproducir audio hay alguna novedad/actualización? Gracias por el buen trabajo!

alepevi, 2017-02-11:
Hola! Enhorabuena por tu trabajo @aikoncwd !!
Habría alguna posibilidad de poder usar Google Photos en Kodi? Sería una pasada poder tener sincronizado este servicio para poder ver las fotos en la TV.
Hablo desde el total desconocimiento, no sé si lo que acabo de comentar es posible o no. Tanto si se puede como si no, mil gracias por tu trabajo, menuda currada!

wincide, 2017-02-12:
Hola!
Para la gente que no tiene tiempo como yo, que exista alguien haciendo estas imágenes es una maravilla!! Gracias!

Sólo añadiría algo para los que usamos aikoncwd sólo como SO y tenemos todas las roms de los juegos , pelis, series , música en un USB:
Quizás ya exista un automatismo pero no lo he encontrado. Cada vez que reinstalo es un poco engorroso tener que ir revisando los paths uno por uno y hacer links simbólicos a los directorios de las roms en el USB, y añadir las fuentes de contenido multimedia paseándonos por los menús.
Sería genial , que en caso de tener el contenido en el usb, existiese un script que automatizase todo este proceso estableciendo los paths contra el USB en ( me lo invento ):
   /media/usb/pelis
   /media/usb/series
   /media/usb/mp3
   /media/usb/RetroPie/roms
   etc ....

Y por desconocimiento porque en esta parte todavía no he podido meterme mucho, ¿ existe algún metabuscador que pongas un canal (u otro contenido) y lo busque en todos esos sources de TV que vienen con aikoncwd? 

Muchas gracias de antemano!

limonetty, 2017-02-13:
Buenas aikoncwd,
Si que vas por faena, hace nada se libero la versión Kodi 17 para Raspbian.
Yo tengo el OpenVpn (server) configurado en tu imagen V6, si lo ves interesante puedo facilitar los 4 pasos para realizar la configuración.
No consume muchos recursos y con un alias detengo el servicio si es necesario

P.D: intente actualizar el gui de  XFCE a pixel pero no logre iniciarlo con el acceso de Kodi (Desktop), no es era importante y lo deje estar.
Un saludo

[114 more replies]