aikoncwd/aikoncwd-rpi-mediacenter #50: error al copiar via samba

jotab24, 2017-02-15:
cuando intento copiar ficheros de más de 2 gigas desde windows 8.1 me dice que falta espacio en el disco pese a que hay espacio de sobra. alguna solucion??

aikoncwd, 2017-02-16:
Las particiones FAT32 solo admiten ficheros de 2Gb como máximo. Quizás el mensaje de error es por ese motivo? Necesitaría más información técnica sobre lo que te ocurre...

jotab24, 2017-02-16:
el usb esta particionado en NTFS de hecho si me bajo un fichero via
transmission da igual el tamaño y anteriormente cuando tenia montado
openelec en la rp3 funcionaba sin problema. la verdad es que no se donde
rascar no se si smb.conf tiene limitado el tamaño

aikoncwd, 2017-02-16:
> no se si smb.conf tiene limitado el tamaño

Que yo sepa no, pero puedes revisarlo por si las moscas

jotab24, 2017-02-21:
he estado haciendo purebas y no encuentro el motivo por el cual va tan lento via WinSCP pasa lo mismo y si comparto una carpeta en windows y me conecto a ella via cifs hago un rsync de un archivo y la velocidad es lentisima.

alguna idea???

dejo copy/paste del rsync via ssh con putty:

root@mediacenter-v6:/media/sharedPort # rsync The\ Walking\ Dead\ 720p\ 7x10.mkv /media/usb0/SERIES/The\ Walking\ Dead/Temporada\ 7/ --progress
The Walking Dead 720p 7x10 [www.newpct1.com].mkv
      2,195,456   0%   22.12kB/s   24:50:58

aikoncwd, 2017-02-21:
Cual es la velocidad de tu microSD? Tienes overclock en el driver de la SD?

jotab24, 2017-02-21:
la verdad es que lo desconozco en esa tarjeta tenia montado anteriormente un openelec y la velocidad de grabación era normal, es más este rsync ataca contra un usb 3.0 por lo que entiendo que la tarjeta microSD no influye

Nideapum, 2017-03-18:
Prueba a copiar los archivos a través del SSH en lugar de usar SMB

closed 2017-04-21 by aikoncwd