aikoncwd/aikoncwd-rpi-mediacenter #51: problemas con quasar en aikoncwd v6

Tuly21, 2017-02-19:
Buenas ante todo gracias por compartir este gran proyecto, y bueno yendo al grano solo queria preguntar que cuando uso pelisalacarta y para reproducir un torrent mediante quasar me ocurre que empieza ha hacer el prebuffering antes de comenzar la reproduccion muy rapido hasta el 40% del prebuffering y posterior la velocidad baja quedandose en 30 kb/s y teniendo pequeños picos de velocidad pero mayormente quedandose a esa velocidad no se el motivo y asi no puedo reproducir las peliculas mediante torrents.

gracias de antemano espero me puedas ayudar

aikoncwd, 2017-02-20:
Que conexión de internet tienes? Tienes la RPi configurada por cable o por Wifi? Tienes los puertos del router abiertos contra la IP de la RPi? Desde que servidor estás visualizando los torrents?

Tuly21, 2017-02-21:
Buenos dias mira ls conexion que tengo es fibra optica 200 megas la rpi 3
modelo b la tengo conectada por cable los puertos que he abierto son los de
quasar, torrents, plexus y demas. Las pelis las intento ver por newpct1.
Ademas decirte estoy intentando que quasar almacene el archivo temporal que
genera en un pendrive pero me va muy lento como te decia y tambien me
ocurre que el torrenter no se puede actualizar.

Gracias de antemano.

aikoncwd, 2017-02-21:
Posiblemente el cuello de botella lo tengas al intentar guardar el torrent en local. Prueba xbmctorrent a ver si te va mejor o elimina la opción de guardar el temporal en el USB.

Tuly21, 2017-02-21:
Y entonces pregunto si no le indico ningun dispositivo externo donde se
guarda el temporal y la otra pregunta es si despues de parar la
reproduccion se elimina ese temporal q imagino estara en la sd en algun
sitio.

Gracias

Tuly21, 2017-02-21:
Otra pregunta porque torrenter no se puede actualizar

aikoncwd, 2017-04-21:
Actualmente ha quedado desfasado, se solucionará en la V7

closed 2017-04-21 by aikoncwd