aikoncwd/aikoncwd-rpi-mediacenter #54: Automontaje en /media/USBxxx

pedropmv, 2017-02-24:
Hola @aikoncwd , he estado mirando y no encuentro la manera de desactivar el automontaje de los USB. Quiero fijarlas yo desde el fstab, pero al arrancar el KODI me las desconecta y me las monta en la carpeta /media

Dreizehn84, 2017-02-25:
Buenas @pedropmv, te copio una respuesta que di a otros usuarios respecto a como configurar el fstab por si le puedo evitar algo de trabajo a @aikoncwd . Configurándolo correctamente creo que no debería haber problema en montar los USB donde quieras. La conversación completa eta en el issue de los cambios en la V7  (#49) , pero te copio lo último que es donde doy más detalles. Ya nos cuentas si te sirve o si te da algún problema.

> @rbf0069 vamos por partes:
> 
> Es normal que tú no tengas la carpeta TOSHIBA ya que es una carpeta que me he creado yo para montar ahí mi disco duro TOSHIBA. Le puse ese nombre simplemente para distinguirlo del otro disco duro que tengo conectado que es SAMSUNG.
> 
> Tú debes crearte una carpeta donde quieras que se te monte cada una de las unidades que tienes. La carpeta la puedes ubicar donde quieras, pero yo la ubiqué dentro de /media para seguir la estructura lógica. Por lo tanto, para empezar, tú podrías crearte las carpetas HDD1 y HDD2 por ejemplo dentro de media, de forma que quedasen /media/HDD1 y /media/HDD2.
> 
> Una vez hecho eso, debemos identificar cada uno de nuestros discos duros por su UUID para no marearnos y poder identificarlos de forma inequívoca en cada arranque. Esto lo haremos fácilmente haciendo uso del comando lsblk -f. Al ejecutarlo, nos devolverá un listado de nuestras particiones, con el punto de montaje que tienen, su etiqueta (si la tienen) y su UUID. El UUID es lo que tendremos que anotar bien para luego configurarlo en el fstab. Este paso que acabamos de hacer lo puedes hacer perfectamente por putty, como tú has dicho.
> 
> Una vez que tenemos nuestras carpetas creadas y los discos identificados por su UUID, lo que tenemos que hacer es editar el fstab con dichos datos. Esto lo puedes hacer también por putty mediante cualquier editor desde la consola, como nano, o también puedes editar directamente el fichero desde windows o bajándotelo mediante ftp y luego subiendo el fichero con los cambios, etc. Ahí ya que cada uno escoja lo que le resulte más cómodo (en mi caso, editarlo con nano es lo más directo).
> 
> El fichero a modificar es el fstab dentro de la carpeta /etc. Para editarlo con nano simplemente tendríamos que poner en consola nano /etc/fstab y ya podríamos añadir las líneas que nos hicieran falta.
> Si no me equivoco, al abrir el fichero nos encontramos 3 líneas que NO HAY QUE TOCAR bajo ningún concepto. Nosotros nos limitaremos a añadir dos líneas más, una para cada uno de nuestros discos.
> 
> ATENCIÓN, antes de proceder a la modificación del fstab, hay que tener en cuenta que cualquier modificación errónea en este fichero puede hacer que nuestra rasp no inicie. No es que nos la hayamos cargado en ese caso, pero sí nos habríamos cargado el fstab y la única forma de solucionarlo sería sacando la microsd de la rasp y conectándola en algún pc con el que podamos leer particiones ext de linux, que en windows, en principio, no podremos. Por lo tanto, mucha atención al modificarlo si no queréis tener trabajo extra.
> 
> Para la modificación, me basaré en la linea de ejemplo que puse para explicarlo:
> 
> UUID=BC2CF6852CF639CA /media/TOSHIBA ntfs defaults,nofail 0 0
> 
> Por cada disco que queremos configurar, añadimos una línea tal como esa, en la que lo único que tenemos que modificar son los UUID que ya anotamos antes tras la ejecución del comando lsblk -f y la ruta de montaje, que en el ejemplo es /media/TOSHIBA y la cambiaríamos por /media/HDD1 por ejemplo.
> 
> Os aconsejo que la ruta de montaje sea sin espacios para que no haya que poner comillas ni nada de eso. Nada en ese comando va con comillas, va todo tal y como aparece en la línea de ejemplo, recordad que unas simples comillas pueden dejarnos colgado el arranque.
> 
> Hay que prestar atención al sistema de archivos del disco duro, que por norma general entiendo que estaréis usando NTFS, pero en caso de no ser así, también tendríais que cambiarlo en esa línea.
> 
> El resto de parámetros aconsejo no tocarlos, a mi esa configuración me va bien y evita que si en algún momento el disco duro no está conectado, se cuelgue el arranque. Aún así, os dejo un enlace a una página con información detallada del fstab por si queréis trastear esos parámetros, pero ya sabemos que cualquier cosa mal configurada nos cascará el arranque.
> 
> https://es.wikipedia.org/wiki/Fstab
> https://wiki.archlinux.org/index.php/Fstab_(Espa%C3%B1ol)
> 
> Para terminar, por supuesto hay que guardar los cambios en la edición del fichero, y mi recomendación es que reinicieis, aunque con el comando mount -a forzareis que se monten las unidades de nuevo.
> 
> Espero que con esta explicación más detallada podáis solucionar los problemas que os está dando el montaje de los discos, pero sigo a disposición del que lo necesite si algo no le ha quedado claro del todo.

aikoncwd, 2017-05-24:
Cerrado. Solucionado por @Dreizehn84

closed 2017-05-24 by aikoncwd