aikoncwd/aikoncwd-rpi-mediacenter #55: Ajustar pantalla hdmi

galastico, 2017-02-26:
Buenas tardes, como se ajusta la pantalla para que la llene?? Sale un recuadro negro y no consigo qie la pantalla se muestre completa.

Como se hace?

Gracias

Dreizehn84, 2017-02-26:
En mi caso simplemente con ajustar el tamaño de pantalla en el televisor pude ajustarlo bien. Si no te es suficiente con eso, en los ajustes del sistema, dentro de Apariencia -> Skin hay una opción que es el zoom. Quizás jugando con él puedas ajustarlo mejor. A ver si te sirve eso. Saludos!

galastico, 2017-02-27:
Hola, es que me pasa en dos televisiones distintas, de marca y todo, y no consigo quitarle el marco. 

Con el zoom probaré a ver

JenXon, 2017-02-27:
@galastico A mi me pasaba lo que comentas, tan solo tienes que coger la sd, meterla en el PC y en el archivo config.txt modificar la línea:

disable_overscan=0              #Disable overscan configuration. Set 1 if you see black lines on TV (default 0)

Tienes que cambiar el 0 después de overscan= por un 1, quedando así:

disable_overscan=1              #Disable overscan configuration. Set 1 if you see black lines on TV (default 0)

También lo puedes hacer por red, te metes en la carpeta /boot y ahí está el archivo config.txt, lo modificas como te he dicho y reinicias. Se te debería solucionar así.

aikoncwd, 2017-04-21:
Problema resuelto correctamente por @JenXon , muchas gracias! :)

closed 2017-04-21 by aikoncwd