aikoncwd/aikoncwd-rpi-mediacenter #57: Problemas con Exodus

javier1056, 2017-03-23:
Buenas tardes. Mi nombre es Javier Ibar y les escribo desde Buenos Aires (Argentina). Tengo funcionando la V6 sobre una Raspberry Pi 3. Tengo el problema que cuando quiera abrir alguna película o serie en Exodus la consola se reinicia por completo cerrando el Kodi volviendo a aparecer las frambuesas del inicio y booteando de nuevo la consola. El problema solo me lo hace con Exodus.
Me podrían ayudar para resolver este problema por favor?

javier1056, 2017-03-23:
La versión de Exodus que tengo instalada es la 3.1.3 si mal no recuerdo

aikoncwd, 2017-04-21:
Es muy posible que el contenido que quieras visualizar por Exodus tenga un bitrate mayor (y no soportado por el hardware de la RPi), por eso se reinicia. Lo mejor que puedes hacer es probar algún contenido de calidad inferior o esperar a la actualización de la V7 para ver si se soluciona.

closed 2017-04-21 by aikoncwd

javier1056, 2017-04-21:
Muchisimas gracias por la respuesta. Te hago una consulta mas. La fecha y hora que me informa el Kodi 16.1 es incorrecta. La cambie en Raspbian pero sigo con el problema en el Kodi. En Kodi tengo configurada la región horaria en Argentina/Buenos Aires.Me podrías ayudar como cambiar la fecha y hora en Kodi por favor?
Saludos y gracias Javier IbarLic. en Sistemas
www.tecnoroot.com.ar
Facebook: Tecno Root