aikoncwd/aikoncwd-rpi-mediacenter #65: Control Remoto  para la Pi

eLPiRRi, 2017-04-23:
Buenas tardes crack,

Para los que somos pez, pero PEZ en linux , llevo pegandome con la imagen tuya v6 para instalarle alguna cosa para controlar remotamente la Pi desde el ordenador windows de casa, o incluso desde el trabajo... por lo que aparte de ver que opciones tengo con la imagen actual, puesto he mirado diversas opciones por la red, pero no consigo instalar ninguna..., o querria consultarte de paso si en la proxima que saques habria opcion o posibilidad que viniese predefinido, alguna app tipo Teamviewer o similar:

https://www.teamviewer.com/es/download/linux/

Es beta pero la hay para la Pi

TeamViewer Host para Raspberry Pi

TeamViewer Host para Raspberry Pi es un servicio de sistema que permite el acceso no supervisado a dispositivos Raspberry Pi para que pueda llevarse a cabo su mantenimiento, control y administración con nuestra solución líder de la industria TeamViewer, conocida por millones de usuarios de todo el mundo.

O si no es esa alguna tipo VNC o similar

Saludos y muchas gracias por tu curro a la comunidad

eLPiRRi, 2017-04-23:
Cierro, 
Puesto que realmente no es una issue el saber conectarme remotamente, seguire investigando por mi cuenta, dejo el post en la V7

https://github.com/aikoncwd/aikoncwd-rpi-mediacenter/issues/49#issuecomment-296440888

Puesto que es problema mi no saber hacerlo o desconocer si tiene alguna herramienta que lo haga en tu imagen V6 , voy a ponerme y a leer los cerrados a ver si doy con algo, y prefiero dejarlo constancia en el otro post, a modo de peticion (aunque se que estan cerradas) el que pueda venir la siguiente con una app con entorno grafico sencilla y al alcance de todos para conectarnos a nuestra Pi 24/7 desde donde queramos

saludos y gracias de nuevo

closed 2017-04-23 by eLPiRRi

guladelnorte, 2017-04-25:
@eLPiRRi una vez que hayas abierto los puertos en el router puedes conectarte mediante consola para gestionar cualquier cosa de la imagen o por web para gestionar retropie y para usar los controles de kodi.

eLPiRRi, 2017-04-25:
Muchas gracias por la contestación @guladelnorte , 

Consegui finalmente hacerlo, gracias a mucha gente que manejan un poco mas linux, por eso cerre este topic y lo puse en el de la V7 para si se considera una herramienta interesante o no, para la gente que no controle mucho de linux poderla tener etc.

apt-get -f install  (tenia muchas cosas mal en tema de librerias, con esto basicamente se soluciono todo e instalo bastantes cosillas, tardo un rato)

Baje TeamViewer Host para Raspberry Pi de https://www.teamviewer.com/es/download/linux/

sudo dpkg -i teamviewer-host_11.0.63329_armhf.deb

Y ya estaría, ahora tambien intentare , que lo hay intentar meter Team Viewer al movil a ver si puedo controlar la Pi desde movil

Saludos