aikoncwd/aikoncwd-rpi-mediacenter #68: Las descargas en transmission van muy lentas

yasmicheta03, 2017-05-16:
Hola,
antes que nada, como siempre decimos todos, la imagen de Aikon es genial! Un trabajo excelente.
Sólo quería comentar que no consigo que los torrents se descarguen a más de 500 kB/s... Los puertos que se mencionan en las instrucciones los tengo abiertos y la línea es de 50MB de bajada. No encuentro que puede ser.
He visto que a alguien más le ha pasado con esta imagen. Ver:
https://www.raspberrypi.org/forums/viewtopic.php?f=76&t=166795

Si alguien me da una pista, lo agradeceré. Gracias!!!

yasmicheta03, 2017-05-16:
Es por culpa del USB!!! A la que le pongo la carpeta de descarga en la SD va genial. De hecho cuando la carpeta está en el USB, va lento hasta conectarme con el navegador...
¿Por qué pasa? ¿Qué se puede hacer?

hacheraw, 2017-05-16:
Soy nuevo con la Raspberry pero CREO que el USB comparte ancho de banda con el puerto Ethernet. Por lo que si usas un pincho USB, tendrás una conexión a internet más lenta.

aikoncwd, 2017-05-24:
Es lo que dice @hacheraw , el bus del ethernet y USB es compartido. Lo recomendable es configurar transmision para descargar en la SD y al finalizar copiar los ficheros al USB (si así lo deseas). Un saludo

closed 2017-05-24 by aikoncwd