aikoncwd/aikoncwd-rpi-mediacenter #71: Illegal Instruction...

ramses-0000, 2017-06-03:
Buenos días,

Estoy intentando instalar aikoncwd v6 en una SD de 8Gb.

Parece que quemarla la quema bien, pero cuando está arrancando en la Raspberry PI 2, entra en un bucle continuo dando un mensaje de "Illegal Instruction...". No sé a qué se debe porque no puedo acceder al log que genera.

He visto que hay gente que le da este mismo error y se han bajado versiones anteriores y no les pasa.

¿Hay algo reportado sobre este error?

¿Podrían decirme si me puedo descargar desde algún sitio la v4 o v5 para probar?

Saludos y gracias por el trabajo.

Jesuspleomax, 2017-06-03:
yo tube ploblema para grabarla y se me soluciono cuando le di formato con gparted a la targeta sd en xt4 suerte amigo

ramses-0000, 2017-06-04:
Pero, supuestamente, cuando copias la imagen con DD a la tarjeta SD, le da formato, ¿no?

Ya he probado con varias tarjetas, tanto desde Windows con Win32 Disk Imagen como desde Linux con DD.

¿Formateando antes la tarjeta SD en formato Ext4 se soluciona el problema?

¿Qué comando exacto usan para volcar la imagen a la tarjeta SD?

Lo curioso es que he probado con otras imágenes de OpenElec y otras, y al intentar iniciar la RPI, ni comienza a cargar el software, y me resulta raro, porque he probado con 4 tarjetas, tanto de 4Gb como de 8Gb, y no sé si todas estarán mal o que no lo esté haciendo bien yo.

Saludos y gracias

ramses-0000, 2017-06-07:
Definitivamente la AikonCWD versión 6 no funciona con las Raspberry 2. He probado una SD con AikonCWD v6 en una Raspberry 3 y arranca perfectamente, y la misma tarjeta la he puesto en 3 Raspberry 2 distintas, y en todas da el mensaje "Illegal Instruction..." y no arranca.

Jesuspleomax, 2017-06-07:
Ami me paso lo mismo y lo solucione de esa manera nose la verdad