aikoncwd/aikoncwd-rpi-mediacenter #75: Plexus-streams Arenavision parser

edumgui, 2017-08-27:
El parser no funciona correctamente desde que algunas operadores españolas han decidido bloquear el dominio .in.

Para solucionarlo, basta editar éste archivo al que se puede acceder vía smb:
`\\MEDIACENTER-V6\kodi\addons\plugin.video.plexus-streams\resources\core\parsers\arenavision\main.py`

Cambiar la línea 21 por:
`base_url = "http://www.arenavision2017.ml"`

Y la línea 118 por:
`arenavision_streams("ArenaVision {0}".format(channelList[index]), base_url + "/" + channelList[index])`

Muchas gracias @aikoncwd por compartir tantas horas de trabajo.

zutik, 2017-08-28:
Gracias por el aporte !

rafanandez, 2017-08-30:
Muchas gracias por el dato @edumgui !

Y si además se sustituye `"agenda"` por `"guide"` en la línea 41, aparece la agenda/schedule en el menú del parseador.

Saludos!