aikoncwd/aikoncwd-rpi-mediacenter #78: Cambio en la resolución de EmulationStation

AntonMaca, 2017-09-14:
Me encuentro ya desesperado con este asunto. Mi televisor es de una marca desconocida, y muuuuy sensible con las resoluciones. 
Con el Config.txt correctamente configurado, Kodi funciona perfecto con esta imagen.
Pero tan pronto entro en EmulationStation, la imagen parpadea. La tele se queda en negro y durante una fracción de segundo, se muestra la pantalla para volver a negro. Y una y otra vez.
He dado mil vueltas por la imagen de AikonCWD, pero no lo encuentro:
¿Dónde se configura, por SSH la resolución de pantalla de EmulationStation?
Se que se puede entrar por el menú de configuración, pero al mostrarse una fracción de segundo, acabas agotado antes de llegar al lugar adecuado.
A ver si alguno me puede ayudar.
Lo que necesito es poder poner EmulationStation en la resolución más baja posible a través de un archivo de configuración por terminal, para luego ya poder ajustar fino con el menú interno.
Saludos.
PD: y felicidades por la imagen. Al margen de este problema (que es culpa de mi tele, no del proyecto), es uno de mis must en las Raspberry!

ismael-dm, 2017-09-15:
Conectas la raspberry a la television por HDMI? Porque si es así, no debería de dar problemas ya que la raspberry adapta la resolución a la pantalla conectada... podría ser un fallo de emulationstation?

AntonMaca, 2017-09-15:
Si, es por hdmi. E insisto que el problema es mi tele, que es muy sensible con la resolución. En Kodi he tenido que poner el hdmi_safe en el config.txt. Y se ve bien y pude ponerle la resolución adecuada por menú. Pero en EmolationStation, no encuentre el archivo de configuración. De ahí mi pregunta, ¿dónde está el archivo de configuración que le dice a ES qué resolución sacar de arranque?