aikoncwd/aikoncwd-rpi-mediacenter #80: Problema al actualizar kodi aikoncwd v6

kojacs, 2017-11-05:
Hola, necesito ayuda... tengo la imagen de aikoncwd v6 y he actualizado kodi 16 a kodi 17 desde los paquetes de retropie y despues de reiniciar no me funciona el acceso de retropie ni desktop desde kodi, ejecuto los accesos y me lleva a un terminal. Alguien sabria solucionar esto por favor?  Muchas gracias

ValkiryXDA, 2017-11-10:
[addon.desktoplauncher.zip](https://github.com/aikoncwd/aikoncwd-rpi-mediacenter/files/1460534/addon.desktoplauncher.zip)

Here you go just place in "/home/*USERNAME*/.kodi/addons" and you should be good to go or here is the command "CD retrolauncher/" then "startx" and you can get back to the desktop again. I'll be making my own iteration of this media center soon. New and Updated version.

Nideapum, 2017-12-17:
Sabéis si sigue funcionando el escritorio, yo no consigo arrancarlo (de echo nunca ha funcionado)
¿Podría ser que solo se puede poner el escritorio por la salida HDMI o eso no tiene nada que ver?

ValkiryXDA, 2017-12-20:
Please make a new issue then. Don't hi-jack someone else's issue.

TopCheff, 2018-03-11:
Yo he tenido que volver a tirar de la imagen.. se me murio kodi y al actualizar no logre funcionarlo.