aikoncwd/aikoncwd-rpi-mediacenter #82: Entrando a root desde otro user por ssh via sudo

spikedboy, 2017-12-03:
Al hacer sudo /bin/su desde otro usuario por ssh, nunca se alcanza el shell, se queda cargando.

Por qué? Se puede solucionar esto?

Un saludo. Y muchas gracias por la distro!

ValkiryXDA, 2017-12-20:
when you connect via SSH you are logging in as root so there is no need for $ su .... This isnt a issue

spikedboy, 2017-12-22:
I disabled root through ssh on purpose. It is a security measure to dificult brute force attacks. The ssh service is exposed to the internet.

ValkiryXDA, 2018-01-02:
thats a bit silly since there is only root account.
Just edit /etc/hosts.deny 
$ sshd: ALL

or set an ip range i.e 10.0.0.0/24 

or

$ sudo iptables -A INPUT -p tcp --dport 22 -j DROP