aikoncwd/aikoncwd-rpi-mediacenter #84: ¿Es compatible con la raspberry pi 3 b+?

Grondoval, 2018-03-27:
Estoy intentando poner la imagen en plan bestia en la raspberry nueva y no carga la imagen quedandose una paletada de distintos colores y además el simbolo de un rayo a la izquierda de la pantalla.

¿Alguién tiene alguna idea o tiene esta raspberry y ha hecho pruebas?

ValkiryXDA, 2018-04-02:
If someone could donate a B+ board to me I can make a build for it

TopCheff, 2018-07-12:
Hola, si lo es! yo lo tengo en una rasp 3 b y perfecto!

ivenegas, 2018-07-13:
Hola. También tengo problemas con raspberry 3 b+ no levanta sistema operativo. Será compatible?

Gracias

TopCheff, 2018-09-05:
Hola, habeis mirado el tema de resolucion ? 
lo que haría yo sería, replanchar la imagen en la SD, usando el windisk 32 o uno que recomienda aikon. y mirar de arrancarla de nuevo. 

en la b + solo han mejorado el procesador, la versión del bluetooh y la entrada eth que ya soporta 300 megas. 

no deberíais tener problema alguno.

Grondoval, 2018-12-07:
Pues nada, no consigo poner esta imagen en la Raspberry Pi 3 B+. Aunque la he grabado con el etcher y con el windisk32. La Raspberry enciende y se oye el ventilador funcionar y las luces encendidas, pero la pantalla no me muestra nada. :(

Kikobeats, 2019-01-10:
me pasa lo mismo :_